oelprinz-org/BlitzboxBL49sp #5: V 0.1.2 speeduino firmware stops working when pulling the USB plug

chaoschris, 2022-04-25:
V 0.1.2 speeduino firmware stops working when pulling the USB plug.
With the USB plug connected all works fine

SW65, 2022-04-25:
Und was haben die als Lösung?

chaoschris, 2022-04-25:
Shutting down the power supply of the Cp2102 converter chip leads to a constant voltage level of 0 V on the RX line of the Atmega2560. This can be fixed temporary with a 10k Pullup from RX to +5V as shown in the picture. Further investigation will follow soon before the next/final pcb revision.
[image]

SW65, 2022-04-25:
Lösung ist halt ohne Zigarettenanzünder mit usb Kabel und kleiner:-)
Liegt nicht an der Firmware?

Melde mich später mal .

Grüße Jörg

chaoschris, 2022-04-25:
Doesn't look like a firmware issue. The original Arduino Mega 2560 design also has two inline resistors with 1K on RX and TX. Will have a deeper look on this later.

chaoschris, 2022-08-26:
solved in V0.1.3

chaoschris, 2022-08-26:
solved in V0.1.3

closed 2022-08-26 by chaoschris