oelprinz-org/BlitzboxBL49sp #6: Would it be possible to get a "how to order" guide?

runesmdk, 2023-02-28:
I've tried uploading the 0.1.4 to jlcpcb, and both the bom and the cpl, but i get so many errors i cannot order it. I must be doing something wrong ofcourse, but i'm no whiz in the world of PCB. Is it possible to get a step-by-step helping hand in ordering it?

chaoschris, 2023-02-28:
done in the speeduino forum:
https://speeduino.com/forum/viewtopic.php?p=61556#p61556

runesmdk, 2023-03-01:
I dont know how i missed that. But what is missing with r21,r23, r40 and ic3, is that only the wideband controller with the cj125 or are other systems affected as well?

chaoschris, 2023-03-01:
You don't missed it, i made it specially for you... ;-) . But this information could be also useful for others, so i decided to publish it on the forum. 
R21 is a pullup and optional. This is the wire which enables the heating of the Lambda sonde via programmable output.
R23 is a Varistor which should be added manually. JLC has nothing in stock to solder on here
R40 a optional BIAS resistor if you decide to use the analog input ADD_LMM with a temperature sensor
IC3 is the Wideband Bosch CJ125 (30522, 30615 or 40103). Must be soldered by hand as JLC cannot provide it for assemby

runesmdk, 2023-03-01:
I am very honered, thank you. How will it heat without the pullup, just always on? 
Where do you buy the CJ125?

chaoschris, 2023-03-01:
The lambda controller will be enabled with a ground signal. you have to program this this feature with a rule in TunerStudio (programmable outputs). You can define two conditions here. one would be RPM greater than XXX and/or engine temperature greater than YY. Both conditions also allow to define an additional time delay.

chaoschris, 2023-03-01:
I am buying the CJ125 always in China or unsolder them from old parts/ECUs......  ;-)
Bosch 30522 is working fine and should be easy to find and also cheap

closed 2023-03-14 by chaoschris