oelprinz-org/BlitzboxBL49sp #10: What board to select in tunerstudio?

runesmdk, 2023-09-13:
There's no blitzbox in the dropdown menu of boards

chaoschris, 2023-09-13:
Hi Rune,
You must use the modified speeduino.ini from 

https://github.com/oelprinz-org/BlitzboxBL49sp/tree/master/software/202305/changes/speeduino.ini
or
https://github.com/oelprinz-org/BlitzboxBL49sp/tree/master/software/202305/reference/speeduino.ini

both are the same

Best regards,
Chris

runesmdk, 2023-09-13:
Thank you. How to upload the firmware when it comes in hex and bin format?
I usually just use speedyloader :D

Den ons. 13. sep. 2023 kl. 09.18 skrev chaoschris ***@***.***

chaoschris, 2023-09-13:
There is no way to upload it with the speedyloader until the blitzbox will be added to the official firmware tree. But also when this is done you need to upload a bootloader in your 2560. So for the moment you need a tool to upload the firmware manually. You can do this with any tool of your choice like usbasp, tinyisp or any cheap arduino board acting as a ISP.

chaoschris, 2023-09-13:
You also need to program the 328p for the lambda controller the same way, so you will really need one of these really cheap programmers. I prefer the usbasp and AVRDUDESS as GUI which is really easy to handle and also works well
https://blog.zakkemble.net/avrdudess-a-gui-for-avrdude/

runesmdk, 2023-09-13:
Okay. I will have to dig into this, never tried it. Have an external wideband controller, so gonna have a look at available pins in the init. When and how does the fuse-programming work, is that while using the isp programmer?

chaoschris, 2023-09-13:
Yes, you can do these settings easily with the same tools you are using to flash the MCU.
Have a look on the Homepage of AVRDUDESS at https://blog.zakkemble.net/avrdudess-a-gui-for-avrdude/  .
There is a picture of the GUI, with this tool it is really easy-going to program the MCUs.
If you don't want to spend 2 or 3 dollars for a programmer you can use any available Arduino Board and use it as a ISP programmer. There is plenty of information on the web how to do this,one example here:
https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoISP

chaoschris, 2023-09-26:
Best practice would be to flash the entire flash with the included bootloader:
ttps://github.com/oelprinz-org/BlitzboxBL49sp/blob/master/software/202305/blitzboxBL49sp_firmware_speeduino-202305_with_bootloader.bin
or
https://github.com/oelprinz-org/BlitzboxBL49sp/blob/master/software/202305/blitzboxBL49sp_firmware_speeduino-202305_with_bootloader.hex

chaoschris, 2023-09-26:
You are talking about the 2560 or the 328P for the lambda?

runesmdk, 2023-09-26:
> You are talking about the 2560 or the 328P for the lambda?
2560. But i managed to get it working, wrote a how-to on how i did on the speeduino forum.

closed 2023-10-11 by chaoschris