oelprinz-org/BlitzboxBL49sp #11: RC/ UAV engine compatibility

professor-jonny, 2023-10-05:
Hi I'm wondering if this could be project could be adapted to accept a 0-5v 50hz PWM input as an electronic throttle and provide a 0-5v 50hz PWM output the same for throttle position control by simply repurposing inputs and outputs without major board changes?

The usage would be used accept a RC throttle control PWM input from the transmitter and provide a PWM output to control the throttle servo.
This would negate the need for TPS as we know the position, it would be otherwise impractical to provide this signal as it would be hard to find of the shelf throttle components suitable for RC/ UAV engine sizes.

In normal cases it would pass through the throttle servo signal direct to the digital servo controlling the throttle, but if comms were lost it would set a pre-determined throttle position for safety or kill the engine.

Would you be at some stage manufacturing / selling this as a kit?

chaoschris, 2023-12-12:
Hello professor-jonny,
unfortunately the project is published as it is and there are no plans to change it in any way. There also wouldn't be no place on the pcb for your planned modifications.
All the needed files for manufacturing are published in the folder "fabrication" and it is easy for everyone to let them build by your preferred manufacturer.
There is also no intention to sell it in any commercial way.

chaoschris, 2023-12-12:
closed as there are no plans to change it in any way

closed 2023-12-12 by chaoschris